OLIMEX/ESP32-GATEWAY #1: Error when open project with Kicad version 4.0.6

hainv1990, 2017-07-13:
Dear Olimex,

I downloaded your hardware design and open project with Kicad v4.0.6 (newest).
But when I try to open kicad_pcb file. I receive an "error loading board. Kicad was unable to open this file... Date of kicad required (or newer): 01/23/2017"
Please help me how i can open this file. What version of kicad you are using.
Thanks alot.

DanKoloff, 2017-07-14:
The error message is pretty self-explanatory. Please use a version of KiCad that was released after 23d of January 2017. Use the latest nightly build of KiCad.

Yes, 4.0.6 is the latest stable release of KiCad but we are a major feedback contributor and we use the latest nightly builds so we can provide the KiCad team with information about behavior of latest release. We often report bugs and issues. We often request features. Not a lot of companies create such complex designs (let alone open-hardware ones) with KiCad and our expertise is important to keep KiCad moving forward.

andrewmarles, 2018-03-21:
Should probably make mention of this in the readme for your project.  Most professional developers use release versions of tools, not experimental.

closed 2018-10-02 by DanKoloff