OLIMEX/ESP32-GATEWAY #7: Rev C board does not work with ETH_PHY_demo out of the box

jcwren, 2018-10-17:
Installed esp-idf v3.1 SDK and the xtensa-exp32-elf toolchain. Cloned the ESP32-GATEWAY project. Built the ETH_PHY_demo with the defaults as specified in the README.me file, and flashed the board. Board is coming up, sees the link is up, but never DHCPs an address. A couple of other Ethernet devices I have correctly DHCP on this same cable.

Green LED is on solid, yellow LED is blinking at 10 to 20 times a second.

Do any changes need to be made to the configuration between the rev B and rev C versions?

```
$ make monitor
MONITOR
--- WARNING: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.usbserial-1430 instead...
--- idf_monitor on /dev/cu.usbserial-1430 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5712
ho 0 tail 12 room 4
load:0x40078000,len:9020
load:0x40080000,len:6064
0x40080000: _iram_start at /Users/jcw/Projects/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x40080330
0x40080330: _KernelExceptionVector at ??:?

I (30) boot: ESP-IDF v3.1 2nd stage bootloader
I (30) boot: compile time 12:33:38
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0bd10 ( 48400) map
I (110) esp_image: segment 1: paddr=0x0001bd38 vaddr=0x3ffb0000 size=0x022d0 (  8912) load
I (114) esp_image: segment 2: paddr=0x0001e010 vaddr=0x3ffb22d0 size=0x00000 (     0) load
I (117) esp_image: segment 3: paddr=0x0001e018 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /Users/jcw/Projects/esp-idf/components/freertos/xtensa_vectors.S:1685

I (126) esp_image: segment 4: paddr=0x0001e420 vaddr=0x40080400 size=0x01bf0 (  7152) load
I (138) esp_image: segment 5: paddr=0x00020018 vaddr=0x400d0018 size=0x27718 (161560) map
0x400d0018: _stext at ??:?

I (200) esp_image: segment 6: paddr=0x00047738 vaddr=0x40081ff0 size=0x06fe0 ( 28640) load
0x40081ff0: esp_clk_apb_freq at /Users/jcw/Projects/esp-idf/components/esp32/clk.c:179

I (213) esp_image: segment 7: paddr=0x0004e720 vaddr=0x400c0000 size=0x00000 (     0) load
I (213) esp_image: segment 8: paddr=0x0004e728 vaddr=0x50000000 size=0x00000 (     0) load
I (225) boot: Loaded app from partition at offset 0x10000
I (225) boot: Disabling RNG early entropy source...
I (231) cpu_start: Pro cpu up.
I (235) cpu_start: Starting app cpu, entry point is 0x40080e9c
0x40080e9c: call_start_cpu1 at /Users/jcw/Projects/esp-idf/components/esp32/cpu_start.c:225

I (0) cpu_start: App cpu up.
I (245) heap_init: Initializing. RAM available for dynamic allocation:
I (252) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (258) heap_init: At 3FFBC3A0 len 00023C60 (143 KiB): DRAM
I (264) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (270) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (277) heap_init: At 40088FD0 len 00017030 (92 KiB): IRAM
I (283) cpu_start: Pro cpu start user code
I (301) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (304) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (304) emac: emac start !!!

I (314) emac: emac reset done
I (454) emac: emac start success !!!
I (4454) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (4454) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHIP:0.0.0.0
I (4454) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPMASK:0.0.0.0
I (4454) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPGW:0.0.0.0
I (4464) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (4474) emac: eth link_up!!!
I (6474) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (6474) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHIP:0.0.0.0
I (6474) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPMASK:0.0.0.0
I (6474) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPGW:0.0.0.0
I (6484) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
<repeats endlessly>
```

jtarazonaf, 2018-12-19:
Same problem

hoene, 2019-01-13:
The current changes in revision D and E to solve this problem are not good either.
GPIO17 and GPIO5 are getting useless and we do not have any replacement on the GPIO header.

DanKoloff, 2019-01-14:
We are, indeed, considering reducing the 4-bit SD to 1-bit SD as in ESP32-EVB and ESP32-PoE. It is in our TODO notes for the next hardware revision of ESP32-GATEWAY. This would free some pins that were used for the Ethernet reset control and clock.

We were forced to have a custom delay circuit in ESP32-EVB since we had no free pins, and we still utilize the clock delay in ESP32-EVB. However, there is no need of such circuit in ESP32-PoE or ESP32-GATEWAY (since we have free GPIO pins). The clock circuit in ESP32-PoE is similar to the circuits of other manufacturers of ESP32 boards with Ethernet, so it should be easier to get it working.

As @hoene noticed in revision D and newer of ESP32-GATEWAY and all revisions of ESP32-PoE the Ethernet requires two extra pins - a power control pin and the Ethernet clock pin. In ESP32-GATEWAY the power control pin is GPIO5 and in ESP32-PoE (the example you linked) the power control pin is GPIO12. The Ethernet clock pin is GPIO17. This is also written in the hardware changes here: https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/HARDWARE

For ESP32-GATEWAY you should edit these definitions in Arduino for ESP32:
 
#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT
#define ETH_PHY_POWER 5

DanKoloff, 2020-07-17:
We made separate folders for older and newer revisions, there is also an Ethernet example for version 4.0 of the toolchain. The main folder now has two sub-folders "Hardware-revisions-B-or-C" and "Newest-board-revisions".

https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/SOFTWARE

closed 2020-07-17 by DanKoloff