OLIMEX/ESP32-GATEWAY #13: ETH Example Never Aquires IP

drdelgado, 2019-05-27:
Plugged ethernet cable into my belkin router, and I continueously see the following message for the ETH example (https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/SOFTWARE/Newest-board-revisions/ETH_PHY_demo):

```
053) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHIP:0.0.0.0
I (55053) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPMASK:0.0.0.0
I (55053) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPGW:0.0.0.0
I (55063) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (57073) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (57073) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHIP:0.0.0.0
I (57073) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPMASK:0.0.0.0
I (57073) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPGW:0.0.0.0
I (57083) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (59093) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (59093) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHIP:0.0.0.0
```

never gets an IP what steps do I need to take to ensure IP connection

drdelgado, 2019-05-27:
@Stanimir-Petev

DanKoloff, 2019-05-28:
It is probably something in your hardware setup or network configuration (firewall, improper settings, routing, etc). We tested the demo for the newer revisions with board from revision E today - it works fine and acquires IP. Please notice that each ESP32-GATEWAY board pass an Ethernet test after manufacturing. This is the log from our test:

```
# make monitor
MONITOR
--- idf_monitor on COM81 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5788
load:0x40078000,len:9176
load:0x40080000,len:6008
0x40080000: _iram_start at C:/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x4008032c
0x4008032c: _KernelExceptionVector at ??:?

I (28) boot: ESP-IDF v3.1.2-dirty 2nd stage bootloader
I (29) boot: compile time 09:13:10
I (29) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0bc94 ( 48276) map
I (109) esp_image: segment 1: paddr=0x0001bcbc vaddr=0x3ffb0000 size=0x022a0 (  8864) load
I (113) esp_image: segment 2: paddr=0x0001df64 vaddr=0x3ffb22a0 size=0x00000 (     0) load
I (116) esp_image: segment 3: paddr=0x0001df6c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at C:/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (125) esp_image: segment 4: paddr=0x0001e374 vaddr=0x40080400 size=0x01c9c (  7324) load
I (137) esp_image: segment 5: paddr=0x00020018 vaddr=0x400d0018 size=0x2788c (161932) map
0x400d0018: _stext at ??:?

I (199) esp_image: segment 6: paddr=0x000478ac vaddr=0x4008209c size=0x07040 ( 28736) load
0x4008209c: _xt_panic at C:/esp/esp-idf/components/freertos/xtensa_vectors.S:361

I (211) esp_image: segment 7: paddr=0x0004e8f4 vaddr=0x400c0000 size=0x00000 (     0) load
I (212) esp_image: segment 8: paddr=0x0004e8fc vaddr=0x50000000 size=0x00000 (     0) load
I (224) boot: Loaded app from partition at offset 0x10000
I (224) boot: Disabling RNG early entropy source...
I (230) cpu_start: Pro cpu up.
I (233) cpu_start: Starting app cpu, entry point is 0x40080ea0
0x40080ea0: call_start_cpu1 at C:/esp/esp-idf/components/esp32/cpu_start.c:231

I (0) cpu_start: App cpu up.
I (244) heap_init: Initializing. RAM available for dynamic allocation:
I (251) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (257) heap_init: At 3FFBC380 len 00023C80 (143 KiB): DRAM
I (263) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (269) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (276) heap_init: At 400890DC len 00016F24 (91 KiB): IRAM
I (282) cpu_start: Pro cpu start user code
I (300) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (321) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (321) emac: emac start !!!

I (321) emac: emac resetting ....
I (321) emac: emac reset done
I (341) emac: emac start success !!!
I (4341) emac: eth link_up!!!
I (4441) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (4441) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHIP:0.0.0.0
I (4441) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPMASK:0.0.0.0
I (4441) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPGW:0.0.0.0
I (4451) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (5311) event: eth ip: 192.168.0.148, mask: 255.255.255.0, gw: 192.168.0.1
I (6461) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
I (6461) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHIP:192.168.0.148
I (6461) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPMASK:255.255.255.0
I (6461) Olimex_ESP32_GATEWAY_REV_B_eth_example: ETHPGW:192.168.0.1
I (6471) Olimex_ESP32_GATEWAY_REV_B_eth_example: ~~~~~~~~~~~
```

drdelgado, 2019-05-28:
@dankoloff what is your network setup here?

DanKoloff, 2019-05-29:
I am connected to TP-Link router with OpenWRT. It looks like this: https://i.imgur.com/TpNQMiH.png