OLIMEX/ESP32-GATEWAY #15: ETH_PHY_demo doesn't build

phibreDE, 2019-08-22:
The network classes have changed in the espressif esp-ide git and the example no longer builds.

`ESP32-GATEWAY/SOFTWARE/Newest-board-revisions/ETH_PHY_demo/main/ethernet_example_main.c:49:10: fatal error: eth_phy/phy_lan8720.h: No such file or directory
 #include "eth_phy/phy_lan8720.h"
`

Could you adapt the examples to the new toolchain?

DanKoloff, 2019-08-22:
Certainly looks like this one: https://github.com/OLIMEX/ESP32-POE/issues/7#issuecomment-523761856

We are aware of the issue that caused it. All ESP32 Ethernet demos of all boards are probably affected.

wlbe4, 2020-06-01:
When can we expect a fix so the examples can compile and run on new toolchain (v4.x)?

DanKoloff, 2020-07-17:
Such example is published here: https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/SOFTWARE/Newest-board-revisions/ESP32_Gateway_Ethernet_v4.0

closed 2020-07-17 by DanKoloff