OLIMEX/ESP32-GATEWAY #20: SD Card example only works with 1-bit mode, not 4-bit

MattEttus, 2020-11-29:
If I choose to use 4-bit mode by commenting out the line:      
     host.flags = SDMMC_HOST_FLAG_1BIT;

Then I get the following errors:

I (0) cpu_start: Starting scheduler on APP CPU.
I (324) example: Initializing SD card
I (324) example: Using SDMMC peripheral
I (334) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (374) sdmmc_sd: sdmmc_check_scr: send_scr returned 0xffffffff
E (374) example: Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.

1-bit mode works fine.  Is something else necessary in order to use 4-bit mode?  The schematic shows all 4 bits being connected.  I am using a Rev. G board.

DanKoloff, 2020-11-29:
This is expected, It is how latest hardware revisions of ESP32-GATEWAY are wired, in order to leave more GPIO pins. You can solder resistors R9, R10, R11 to re-enable 4-bit mode of operation.

Refer to the revision changes notice for revision F here: https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/HARDWARE

MattEttus, 2020-11-30:
Thanks, I missed that in the doc.  I'll ad din zero ohm Rs.

closed 2020-11-30 by MattEttus