OLIMEX/ESP32-GATEWAY #23: Wire1 not working in revG

Razzorback225, 2021-10-07:
Hi,

I've tried several times to get Wire1 working as the default pins for I2C are not available but could not achieve this.

I tried to setup Wire1 with these pins :

16 for SDA, 17 for SCL
36 for SDA, 35 for SCL
15 for SDA, 2 for SCL

And function used was 'Wire1.begin(SDA, SCL, 100000)'

All the tests have been done without SD card mounted or Ethernet.

Can I have a bit of help on this please?

DanKoloff, 2021-10-07:
Refer to this: https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/resources/ESP32-GATEWAY-GPIOs-Rev.F-up.pdf and also refer to the schematic.

I would avoid pins 16 and 17 since they are used for the Ethernet chip. No matter if you have cable attached the pins are in use.

Notice that GPI35 and GPI36 and GPI39 are only inputs for the board (hence why are they named GPI instead of GPIO).

Use GPIO4, GPIO12, GPIO13, GPIO32, GPI35, GPI36, GPI39.

DanKoloff, 2022-02-15:
Closing due to inactivity.

closed 2022-02-15 by DanKoloff