OLIMEX/ESP32-GATEWAY #29: Can no longer flash after a reset - Failed to connect to ESP32: No serial data received.

malandles, 2023-04-12:
## Specifications
- Board: ESP32-GATEWAY-EA
- ESP-IDF: v4.4.4

## Issue
Flashed the [hello-world example](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/get-started/hello_world) from the ESP-IDF repository.

Had to press `BUT1` to enable download mode, then `RST1` to reset the board, then I could monitor successfully the hello world:
```
Hello world!
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision v3.0, 4MB external flash
Minimum free heap size: 301540 bytes
```

But then, I am unable to put back the download mode, each time I try to flash, I get this error:
```
esptool.py -p /dev/tty.usbserial-1460 -b 115200 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader/bootloader.bin 0x10000 ethernet_basic.bin 0x8000 partition_table/partition-table.bin 

esptool.py v3.3.2
Serial port /dev/tty.usbserial-1460
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

Did I overwrite Olimex's bootloader? Is there a manipulation that can be done to reset to factory?

DanKoloff, 2023-04-13:
Probably a driver issue with CH340T. Make sure to reinstall the CH340 drivers.

You said that you held button BUT1 but that is unnecessary and probably did nothing, Olimex ESP32 boards enter programming mode automatically. Did you first configure the button BUT1 for manual boot operation? By default BUT1 acts only as a regular button connected to a GPIO. To be able to use it for manual boot you need to place 0 Ohm resistor on R46 pads, refer to the right side of the schematic where buttons are.

DanKoloff, 2023-04-14:
The button didn't do anything for sure, it is likely the restart that influenced the successful entry into bootloader mode. Also I am not sure but it might have something to do with these flags in your command line "--before default_reset --after hard_reset ", double check if your command parameters are required. Probably some timeout occurs and if you can replicate the restart timing you will probably be capable of programming the board again.

If nothing else works you should try connecting the boot button (placing 0 ohm resistor on the missing R46, it is 0603 size) but first:

- Double check your command line parameters;
- Try reinstalling the CH340 drivers;
- Try another USB port and cable, if using USB hub, consider testing without it;
- Try another version of esptool or maybe try with another tool to program the board like ESP32 for Arduino.

Some combinations of versions of esptool and CH340 drivers are known to cause timeouts. I'd also recommend searching online for similar timeouts between CH340 and esptool, check these:

https://github.com/espressif/esptool/issues/443
https://github.com/espressif/esptool/issues/592
https://www.olimex.com/forum/index.php?topic=8214.0

malandles, 2023-05-01:
- ❌  Double check your command line parameters; 
  - Removed `--before` and `--after`, did not work
  - Replaced `--before default_reset` by `--before no_reset`, did not work
- ❌  Try reinstalling the CH340 drivers, did not work
- ❌  Try another USB port and cable, if using USB hub, consider testing without it; 
   - Used multiple different cables, all had the same behaviour
   - Used multiple different ports (USB-C -> MicroUSB or USB-A -> MicroUSB), all had the same behaviour
   - Not using a USB hub
- ❌ Exact same error with ESP32 for Arduino: `A fatal error occurred: Failed to connect to ESP32: No serial data received.`

I will try to manually connecting to the boot.

closed 2023-05-01 by malandles